SlicerIGT/SlicerIGT #22: (PLUSRemote) Multiple recording device support

ihnorton, 2015-03-20:
(opening in SlicerIGT to replace https://github.com/SlicerIGT/PlusRemote/issues/7)

I think I actually need this because PLUS does not seem to support either mixing video streams or recording multiple video streams with a single CaptureDevice. I will work on it.

ihnorton, 2015-03-20:
I made a small test change:
```
  def onStartRecording(self):
	for dev,fname in [["CaptureDeviceLeft","Recording-L-"], ["CaptureDeviceRight", "Recording-R-"]]:
		self.logic.startRecording(self.linkInputSelector.currentNode().GetID(),
								  dev,
							      self.logic.addTimestampToFilename(fname),
								  self.printCommandResponse)
```

First Start command works:
```
Command StartRecording [7]: Success
VirtualStreamCapture (CaptureDeviceLeft) StartRecording successful
```

Unfortunately the second Start command fails with:
```
Warning: In ..\..\..\SlicerIGT\OpenIGTLinkRemote\Logic\vtkSlicerOpenIGTLinkRemoteLogic.cxx, line 192
vtkSlicerOpenIGTLinkRemoteLogic (0000000011931A00): vtkSlicerOpenIGTLinkRemoteLogic::SendCommand failed: command is already in progress
```

Do I need to do something to flush the first command before calling?

ihnorton, 2015-03-20:
Nevermind. I see each command is a `vtkSlicerOpenIGTLinkCommand` instance that is async until some completion event.

lassoan, 2017-07-03:
It seems there is nothing to fix.

closed 2017-07-03 by lassoan