SlicerIGT/SlicerIGT #23: Use the common simplified fiducial list widget in Fiducial registration wizard

lassoan, 2015-05-15:
The fiducial widget is moved from Fiducial registration wizard to the Slicer core to make it usable for other modules, too.

Need to update the Fiducial registration wizard module to use this common implementation.

cpinter, 2015-05-19:
Progress is tracked in the following repository:
https://subversion.assembla.com/svn/perklabsandbox/trunk/SimpleMarkupsWidget/

The final version will be committed to Slicer core in Modules/Markups/Widgets, and the PerkLabSandBox folder will be removed.

tavaughan, 2017-06-22:
The Fiducial Registration Wizard module uses the qSlicerSimpleMarkupsWidget. Is this correct? Is the issue resolved?

lassoan, 2017-06-22:
Yes, it's fixed now.

closed 2017-06-22 by lassoan