SlicerIGT/SlicerIGT #29: Disable WW/WL change while dropping contour points

lassoan, 2015-07-22:
The WW/WL of the image is often accidentally changed when adding (slightly dragging after dropping) fiducial on the image, especially when adding multiple points quickly, by using a touch screen.

Disable WW/WL change on the image while contouring. Create helper functions that makes it easy to do this in other states, too.

Request a fix in Slicer (from Nicole or Steve) so that drag-and-drop in new markup adding mode won't change the image brightness/contrast.

tavaughan, 2017-06-22:
Does this issue pertain to LumpNav, MarkupsToModel, or another module? Has it been resolved?

tavaughan, 2017-06-22:
Actually, I'm pretty sure now this is resolved:
1) Click to Place Fiducials
2) Click to put it in persistent mode
3) Click and drag on an image in the slice view
=> No change in intensity

closed 2017-06-22 by tavaughan

lassoan, 2017-06-22:
We can now disable certain interactions in a slice view. For example, disable WW/WL change:

    interactorStyle = slicer.app.layoutManager().sliceWidget('Red').sliceView().sliceViewInteractorStyle()
    interactorStyle.SetActionEnabled(interactorStyle.AdjustWindowLevelBackground, False)
    interactorStyle.SetActionEnabled(interactorStyle.AdjustWindowLevelForeground, False)

But I could not reproduce the problem either, so we can add this lock-down if there is a confirmed need.