SlicerIGT/SlicerIGT #30: PlusRemote reusing GUI elements

che85, 2015-07-22:
I think it would be a good idea to make not only the logic reusable but also the widget "parts", so that one can easily reuse these GUI elements instead of reimplementing them. 

I is just one thought but something like:

<code>
recordingWidget = slicer.modules.PlusRemoteWidget.getRecordingWidget()
</code>

<code>
myLayout.addWidget(slicer.modules.PlusRemoteWidget.startStopRecordingButton)
</code>

Anyway I think that is a general issue which fits for a lot of modules.

When you need to access the logic, so the PlusRemoteWidget is also available.

Does anything argue against doing it this way?

ungi, 2015-07-22:
Good point. But I think PlusRemote was meant to be an example how to use OpenIGTLinkRemote from python. To create reusable widgets, I think PlusRemote would need to be rewritten in C++. When I last checked, using a python module from another python module was very much like hacking.
I'm not sure but I think the GuideletLib is supposed to provide you with such functionality when you base your module on that.

lassoan, 2015-07-22:
Yes, there are two main options:
* Custom Qt widgets in C++; similarly to existing MRML widgets, such as node selector or markup list
* Python functions that add the widget to a window and processing logic (all implemented in a helper class or base class; similarly to the reload&test widget in the scripted loadable module base class

We can start implementing everything in Python and when we see that the complexity is not manageable or if we want to use a widget in C++ then we implement that widget in C++.

che85, 2015-11-03:
Everything has been clarified. Closing

closed 2015-11-03 by che85