SlicerIGT/SlicerIGT #32: PlusRemote: live reconstruction fails

che85, 2015-07-24:
```
Traceback (most recent call last):
  File "/Applications/Slicer 2.app/Contents/Extensions-24459/SlicerIGT/lib/Slicer-4.4/qt-scripted-modules/PlusRemote.py", line 776, in onStartStopLiveReconstructionButtonClicked
    self.onStartReconstruction()
  File "/Applications/Slicer 2.app/Contents/Extensions-24459/SlicerIGT/lib/Slicer-4.4/qt-scripted-modules/PlusRemote.py", line 931, in onStartReconstruction
    self.logic.startVolumeReconstuction(self.linkInputSelector.currentNode().GetID(), self.volumeReconstructorIDSelector.currentText, self.outputSpacingValue, self.outputOriginValue, self.outputExtentValue, self.printCommandResponse, self.getLiveReconstructionOutputFilename(), self.liveOutputVolumeDeviceBox.text)
  File "/Applications/Slicer 2.app/Contents/Extensions-24459/SlicerIGT/lib/Slicer-4.4/qt-scripted-modules/PlusRemote.py", line 1418, in startVolumeReconstuction
    self.cmdStartVolumeReconstruction.SetCommandAttribute('OutputSpacing', '%f %f %f' % tuple(outputSpacing))
TypeError: 'int' object is not iterable
```

lassoan, 2015-07-24:
How did you get this error? Which Slicer version? Maybe you entered a single value where 3 values were expected?

che85, 2015-07-24:
I think, that the problem is, that in onStartReconstruction, startVolumeReconstuction is called with parameter self.outputSpacingValue which is initialized with 0 in __init__

che85, 2015-07-27:
Is a scout scan recording essential for creating a live reconstruction? 

When I first execute the scout scan, then it works as expected

lassoan, 2015-07-28:
Yes, scout scan is needed. For live reconstruction you need to specify image position, orientation, spacing, and extent in advance. Probably the reconstruct button should be disabled if no scout scan is performed.

che85, 2015-07-28:
Yes that makes sense. You also can think about running the scout scan shortly when for live reconstruction no ROI, Extent and Origin has been found.

che85, 2015-11-03:
Since everything has been clarified, I will close this issue

closed 2015-11-03 by che85