SlicerIGT/SlicerIGT #33: Guidelet Class Design

che85, 2015-07-30:
Hi,
I wonder why we are using a class hierarchy like the following:
[image]

Instead of using one like:

[image]

I mean all general GUI elements shall be implemented in GuideletWidget. All specific GUI elements shall be added by UltraSound which also could have its own logic class.  

I don't really see the need of a separate Guidelet class.

Maybe you can explain, why you designed it like that?

lassoan, 2015-07-30:
GuideletWidget is the module widget that appears in the module docking window. Guidelet is a separate docking window. So, they are two separate widgets that can be nicely separated, so they should be in separate classes to have high cohesion and low coupling between classes.

che85, 2015-07-30:
Ok I got it.

But what about the logic? It should belong the the Guidelet then and not to the GuideletWidget, right? Or is the GuideletLogic supposed to be used by both?

lassoan, 2015-07-30:
The GuideletLogic is used by both widgets (Guidelet and GuideletWidget).

closed 2015-07-30 by che85