SlicerIGT/SlicerIGT #35: CollectFiducials should reset counter when a new Markup list is created

ihnorton, 2015-08-12:
Right now the points are consecutively numbered forever. Should reset counter to 0 when the fiducial list node is (re)set in logic.

lassoan, 2015-08-13:
Resetting the counter when the fiducial is selected may have unintended consequences: when loading the scene or just temporarily switching to another fiducial node may reset the counter. Instead, maybe add a button to reset the counter.

tavaughan, 2017-10-05:
I've added a spinbox to expose the counter variable. The counter will now also be saved in a Collect Fiducials MRML node, so its state will be preserved on reloading a scene.

Is this sufficient to close this issue?

lassoan, 2017-10-05:
Yes, I think so. Thank you.

closed 2017-10-23 by tavaughan