SlicerIGT/SlicerIGT #44: Add module for combining and inverting transforms

lassoan, 2015-09-30:
Often it is useful to have an inverted transform in the tree (for example, we receive ReferenceToTracker from OpenIGTLink but we would like to apply TrackerToReference transform to objects).

Also, it would be useful to be able to generate transforms by a sequence of concatenation/inversion.

The module could have a table where we define from/to coordinate system names for transforms (a heuristic can be used for obtaining that from the name of the transform) and another table that would contain computed transforms.

tavaughan, 2017-06-22:
This functionality will be available in Transform Fusion.

lassoan, 2017-06-22:
Great! Maybe then TransformFusion can be renamed to something like TransformProcessor or Process Transforms.

tavaughan, 2017-10-23:
See Pull Request #142. If we want to rename the module, then can we create a new ticket?

closed 2017-10-25 by tavaughan