SlicerIGT/SlicerIGT #52: Error message missing when points are identical in one list

lassoan, 2015-12-03:
From @ungi 

How to reproduce.

Create two fiducial lists.
Use the fiducial placement mouse mode to create three random points in the first list.
Use the Place fiducials function to put three points in the origin, using an identity transform.
Observe the result. It says success! It should say "Error: some points are collinear." Or "Error: some points are not independent. Try to define your fiducial points again."

lassoan, 2015-12-03:
From @mholden8 

The functionality exists, but was commented out in the commit @d9aceae due to compatibility issues with VTK6. Uncommenting these lines and fixing the errors should probably do the trick.

tavaughan, 2017-06-07:
Original title: Error message missing when points are identical in one list

Renamed the issue because the problem is broader than when points are identical

closed 2017-10-30 by tavaughan