SlicerIGT/SlicerIGT #53: Place fiducial gesture recognition

lassoan, 2015-12-03:
From @ungi 

When the selected "StylusTip" is in a pivoting motion for some amount of time, a fiducial should be automatically placed. Probably a checkbox would be also needed to make this feature optional. (Checkbox next to Place button.)

tavaughan, 2017-11-03:
@lassoan Is this not a feature of PLUS now?

lassoan, 2017-11-03:
Yes, it is available in PLUS (in vtkPlusLandmarkDetectionAlgo). We would "just" need to make it available in SlicerIGT.

To do that, probably we should build Plus as part of SlicerIGT (similarly how Plastimatch is built as part of SlicerRT). This would also give us access to volume reconstruction, scan conversion, bone enhancement, ultrasound simulation, transform compositing, etc. We've been delaying this step for a long time, but maybe we should just do it.

@ungi @cpinter What do you think?