SlicerIGT/SlicerIGT #54: Add fiducial matching algorithm

lassoan, 2015-12-03:
From @mholden8 

There should be a matching algorithm (optional) added to automatically determine the correspondence between fiducial points in the two fiducial lists. Possibly should be made to work when one fiducial list only contains a subset of points.

Add brute force algorithm for fiducial matching.
Refine brute force algorithm to improve performance (perhaps using a branch and bound type algorithm).
Make usable for when there is only a subset of points.
For most practical applications, an polynomial time approximate closed-form solution won't be necessary. A brute force algorithm should be sufficient for 6-10 fiducials.

If there is any ambiguity in the matching, report the best result and give some indicator that there is possible ambiguity in the matching.

tavaughan, 2017-06-06:
Does this need to work with the "Similarity" and "Warping" transforms?

lassoan, 2017-06-06:
I think the matching should always use rigid transform. If you select a non-rigid transform you could still perform point matching and outlier rejection are completed using rigid assumption and then use the resulting pointset for non-rigid registration.

If this does not work well then we can simply disable auto-match for non-rigid transforms.

tavaughan, 2017-06-07:
Might it be worth putting the point correspondence-finding algorithm into its own file/logic? If so, what would be the way to do it? Are there examples in SlicerIGT?

lassoan, 2017-06-07:
I would put it in the module logic first. If the file grows bigger than a few ten KB then it's definitely better to split out the point matching algorithm into a separate file. There are several C++ modules that have multiple files in the module logic directory.

tavaughan, 2017-06-14:
There are a few TODO's in the code. Addressing them would make the implementation cleaner so I'm going to do those first.

I should rephrase my earlier question - Are there *preferred* examples in SlicerIGT that I could refer to for multiple logic files? I will likely want to do this later on for MarkupsToModel.

lassoan, 2017-06-14:
There is nothing special about having multiple files in the logic, just add all files to CMakeLists.txt in the Logic directory. See for example OpenIGTLinkRemote\Logic.

tavaughan, 2017-06-14:
@mholden8 @lassoan What is the purpose of the InputDataModifiedEvent? In other classes I see it invoked only when the input MRML nodes (markups, transforms, etc) are changed. In this class I see it is also invoked when modes of operation are changed (update manually/automatically, registration rigid/similarity/warping). Is there a reason for this difference?

tavaughan, 2017-06-17:
Is reflection ever an operation that needs to be taken into consideration?

lassoan, 2017-06-17:
We often rely on asymmetric patterns that provide a unique fit. Allowing reflection (mirroring) may lead to ambiguity. So, I would not allow reflection.

tavaughan, 2017-06-22:
I have a brute-force algorithm that works for two small point sets of equal size. The complexity is N!, but the performance seems to be okay as long as the number of points in each set <= 8.

There are several things that can still be improved:
- Check geometry to make sure there is no ambiguity (including reflection)
- Allow registration for point sets of non-equal size
- Implement a faster algorithm

Would this be suitable for merging? Since this is a new feature, I think Gabor will want to know about it before it is pushed. I'll send him an email.

lassoan, 2017-06-22:
> Allow registration for point sets of non-equal size

This is very important to implement. The most common error is that we miss a landmark but don't know which one so we have to record all the points again.

tavaughan, 2017-10-23:
I'm closing this issue because the point-matching algorithm has been 'added'.

I'll create a new issue for warning output when there is ambiguity.

It isn't clear yet if the algorithm needs to be optimized. If it turns out that it *does* need to be faster, then another ticket should be created (referencing this one).

closed 2017-10-23 by tavaughan