SlicerIGT/SlicerIGT #58: GUI expands when there is a long transform name

lassoan, 2015-12-03:
from @ungi 

How to reproduce:
1. Create a transform with a name at least 200 characters.
2. Switch to PivotCalibration module.
At this point the module panel expands, and takes up space from the 2D/3D viewers. The panel cannot even be set back to a convenient side by dragging its right edge.

The module should not expand at all. The node selector should behave as in the Transforms module. That the name is converted to shorter using "..." in the middle.

tavaughan, 2017-10-24:
(ignore reference from commit above, it was a typo)

tavaughan, 2017-10-26:
This problem is not specific to SlicerIGT. For instance:

1. Navigate to FiducialToModelRegistration
2. Create an output transform with a silly long name like (Create Linear Transform As...):
LongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLongLong
3. Navigate to Transforms module.

The transforms module will expand to take up space from the viewers.

@lassoan where would be a suitable place to report this?

tavaughan, 2017-10-26:
I think I found the culprit, see
http://doc.qt.io/qt-4.8/qcombobox.html#sizeAdjustPolicy-prop

Default is to expand to fit the contents on first showing. I'm trying a different setting (AdjustToMinimumContentsLengthWithIcon)

lassoan, 2017-10-26:
The issue is that there is no elide mode for QComboBox (https://forum.qt.io/topic/3975/how-to-set-qcombobox-text-elide-mode). That's why I think CTK has added ctkComboBox. qMRMLNodeComboBox should be able to support elide mode, too, but maybe you have to enable it.

lassoan, 2017-10-26:
I can confirm that slicer.qMRMLNodeComboBox() can elide, too:

```
c=slicer.qMRMLNodeComboBox()
c.setMRMLScene(slicer.mrmlScene)
c.show()
```

tavaughan, 2018-08-21:
Just to follow up, it seems that setMRMLScene() enables or disabled the elide. So we just have to make sure we call it whenever we have a qMRMLNodeComboBox.