SlicerIGT/SlicerIGT #63: Save/load settings with scene

lassoan, 2015-12-03:
from @ungi 

Parameters and advanced parameters should be restored with a scene is reloaded.

tavaughan, 2017-10-30:
On loading a scene with every setting:

```
qSlicerSimpleMarkupsWidget::updateWidget failed: Markups module logic or scene is invalid
qSlicerSimpleMarkupsWidget::updateWidget failed: Markups module logic or scene is invalid
void qSlicerMarkupsPlaceWidget::setPlaceModeEnabled(bool)  activate failed: Markups module logic, scene, or interaction node is invalid 
void qSlicerMarkupsPlaceWidget::setPlaceModeEnabled(bool)  activate failed: Markups module logic, scene, or interaction node is invalid 
void qSlicerMarkupsPlaceWidget::setPlaceModeEnabled(bool)  activate failed: Markups module logic, scene, or interaction node is invalid 
void qSlicerMarkupsPlaceWidget::setPlaceModeEnabled(bool)  activate failed: Markups module logic, scene, or interaction node is invalid 
```

In addition:
- Fiducial matching was changed from 'Manual' to 'Auto'
- The transform I had in the "Preview Transform" was duplicated (there are now two copies of it)

tavaughan, 2017-10-31:
@lassoan there is one outstanding issue. When I reload the scene, the module will not remember which nodes I wanted to have previews for (no checkboxes next to those nodes in the drop-down list). The hidden preview nodes still exist, but their associations to the module and to the original node are gone. Is this an issue? If so, what would be a good way to go about resolving this?

tavaughan, 2018-07-05:
@lassoan

lassoan, 2018-07-06:
If not saving the preview nodes with the scene is really easy to implement then we can do that (disable save with scene flag for the data node and display node). If not saving the preview nodes is not completely trivial then save the preview nodes as node references.