SlicerIGT/SlicerIGT #65: Record orientation too (not just position)

lassoan, 2015-12-03:
from @ungi 

It could be useful for showing oriented glyphs. Although currently the markups module does not orient glyphs based on orientation information.

tavaughan, 2017-06-22:
Would the best approach be to generate glyphs from sequences? Then sequences could be recorded instead of markups?

lassoan, 2017-06-22:
Yes, that's a good idea. Now that we have Sequences, we should retire CollectFiducials and have a module that creates a model (using glyphs) or fiducial list from a transform sequence (probably by selecting a transform node - which may be a regular transform node or a proxy node; and a sequence browser node; and maybe also some start/end frame).

tavaughan, 2017-10-03:
Perhaps this should be marked as invalid?

lassoan, 2017-10-04:
Agreed.

closed 2017-10-04 by lassoan