SlicerIGT/SlicerIGT #79: MarkupsToModel crashes on Linux

aniqah, 2016-05-12:
Slicer crashes upon opening the MarkupsToModel module on Linux. This problem does not occur with other modules, just this one. Sometimes the module can be loaded, but will crash Slicer upon selecting the "Curve" mode.

aniqah, 2016-05-12:
Here's the problem:
```
Thread 1 "SlicerApp-real" received signal SIGSEGV, Segmentation fault.
strlen () at ../sysdeps/x86_64/strlen.S:106
106	../sysdeps/x86_64/strlen.S: No such file or directory.
```
I can't find the `sysdeps` directory let alone `strlen.S`, and don't think it exists.

ihnorton, 2016-05-12:
If you are on linux you probably would need to install the standard library `-dev` packages to get those headers/asm files. However, those libraries are *probably** not broken. Look further up the stack, most common cause is someone passing a null pointer.

(* gcc 5.2 has some huge ABI breakage so could be a problem if you mixed compiler versions/vendors... but eliminate other possibilities first)

aniqah, 2016-05-16:
The problem was on my side, I needed to rebuild the source. Sorry about that and thanks!

closed 2016-05-16 by aniqah