SlicerIGT/SlicerIGT #81: Spin calibration button not active

ungi, 2016-06-01:
Currently the spin calibration becomes active only when pivot calibration is done with the same input selections. But when we want to calibrate a stylus using recorded sequences, we want to change the input to the sequence that contains spin calibration data. But at this point the spin calibration button becomes inactive. I think the spin calibration button should always be active, and just suggest users to do pivot calibration first.

mholden8, 2016-06-01:
What specifically is you application and workflow? Do you not do pivot calibration with a recorded sequence first?

There are two reasons why it is disallowed to do spin calibration without pivot calibration:
1. Spin calibration does not really make sense without pivot calibration.
2. The spin calibration depends on the pivot calibration to determine the direction of the stylus shaft (i.e. +z axis vs. -z axis).

I suppose the user could have already done a separate pivot calibration beforehand, so perhaps the first reason is not too much of an issue. To overcome the second issue, users not doing pivot calibration would have to manually specify the +/- direction of the needle axis.

ungi, 2016-06-01:
I have two sequence files recorded. One contains pivoting motion, the other contains spinning motion. When I load these two sequences in Slicer, two StylusToReference transforms are created (as instances of separate sequences). I select the one for pivot calibration as input. After successful pivot calibration, the spin calibration button becomes active. Then I select the other transform that is doing the spin calibration, but as soon as I select it in the combo box, the spin calibration becomes disabled.

Putting both pivot calibration and spin calibration in the same recorded sequence is not a good solution, because we may accidentally run into the spin phase while the pivot calibration is still running.

mholden8, 2016-06-01:
Ok, that makes sense. I suppose the current setup prohibits doing just a spin calibration also (where the pivot calibration is already done separately).

I propose the following:
1. Display a warning if a spin calibration is started without a previous pivot calibration.
2. Add an option to "flip" the spin calibration (in case the +/- direction is improperly determined).

What do you think?

ungi, 2016-06-01:
OK. Thanks!

lassoan, 2016-06-01:
It would be nice if the order of pivot and spin calibration would be completely arbitrary. We could let both calibrations button enabled all the time then.

I don't think a flip button is needed, as we without the tip position the tool position would be quite random anyway. It's enough if we flip automatically when pivot calibration is performed.

ungi, 2016-06-02:
These are not exclusive ideas, all can be implemented.
If the translation part of the output transform is not zero, then spin calibration would work as it works now. It will assume that the shaft is on the Tool side.
Andras' idea can be added: When pivot calibration is performed and the rotation part of the output matrix is not zero, then pivot calibration assumes that the shaft should be on the Tool side, and flips the orientation accordingly.
If calibrations (pivot or spin) is started with a fresh output transform, then it doesn't assume anything and just picks a random side and fills in the rotation or translation part of the output transform.
I don't have a preference on flip button. I probably would not use it, if we have the above features.

mholden8, 2016-06-02:
Thinking about this more, what you propose makes complete sense - these are more or less the same thing... Checking after either spin or pivot calibration if the flipping criterion is satisfied (the same flipping criterion as currently) would be sufficient. And if either calibration hasn't already been done, the flipping would be arbitrary.

commit 579afe2f (2016-06-19): Merge pull request #83 from mholden8/master

Fixes #81: Allowing pivot and spin calibration in arbitrary order.
closed 2016-06-19 by lassoan