SlicerIGT/SlicerIGT #85: Crash when launching guidelet

mholden8, 2016-06-16:
Reproduction steps:
1.	Install SlicerIGT extension.
2.	Add UsNeedleTutor guidelet to Slicer.
3.	Start the guidelet.

This causes Slicer to crash.

A few observations:
-	This happens on the latest nightly, as well as the nightly from about two weeks ago
-	We can reproduce on both my computer and another computer
-	The same crash occurs for the LumpNav guidelet
-	The error seems to occur at the line: https://github.com/SlicerIGT/SlicerIGT/blob/master/Guidelet/GuideletLib/UltraSound.py#L126
-	It may have to do with event triggering, since when we went through it step-by-step with the Python debugger, it sometimes worked

lassoan, 2016-06-16:
The best would be to run it in a debug-mode-built Slicer. The stack trace probably would show where the problem is. In the specific line I've noticed that a hardcoded 6 value was used instead of a constant - it would be nice to change this, but probably doesn't cause the crash.

ungi, 2016-06-17:
For those who are reading this issue and don't know how to work on their guidelet-based code until this is fixed. Download Slicer from the June 1st page (The installer has the date: 2016-05-31) and install SlicerIGT extension in that. The namespaces in that version already follow the current style, but this bug seems to be not there yet. This is the Slicer installer for Windows:
http://slicer.cdash.org/viewFiles.php?buildid=837392

commit c6ef3fe4 (2016-06-19): fixed #85: Fixed guidelet crash

A recent reorganization of vtkMRMLSliceNode broke sliceNode->SetSliceToRAS method. Removed the method from the core to avoid application crash.

The transformation matrix can be set by using sliceNode->GetSliceToRAS()->DeepCopy().
closed 2016-06-19 by lassoan