SlicerIGT/SlicerIGT #92: MarkupsToModel auto update checkbox on makes using the module error-prone

cpinter, 2016-07-27:
It's quite unexpected (and unprecedented in Slicer) that simply switching to a module initiates processing. 
Also I lost data by removing the automatically created model while auto-update was still on.
I think it would be better if the auto-update combobox was off by default.

lassoan, 2016-07-27:
@tavaughan has been working on a complete GUI rework - it's almost-ready for integration (https://github.com/SlicerIGT/SlicerIGT/pull/90). We should test the behavior again after his changes have been integrated.

Probably the issue is that "None" option is not enabled in the output node selector and so it selects the first model node in the scene.

cpinter, 2016-07-27:
OK thanks! I wasn't aware of the UI rework.

Why is auto-update on by default? I think the problem is not (only) the None option, but more the auto-update, as I suggest in the ticket title.

lassoan, 2016-07-27:
Auto-update is good, because that's what you need most of the time. Auto-update is enabled by default in Fiducial registration wizard, too, and nobody complained. The problem is that the module randomly selects a node as you enter the module and you don't have a chance to turn off auto-update or choose a different output node.

cpinter, 2016-07-27:
Adding the None options would help for sure.

I think it is very unusual in Slicer that by changing selection in a node combobox immediately changes the content of the node without explicit action (may it be enabling auto-update or clicking manual update). I didn't see any module operating like this before outside SlicerIGT.

lassoan, 2016-07-27:
Yes, SlicerIGT is "unusual" because it mainly operates on nodes that are changing in real-time, that's why auto-update is the default for most modules (but it can be disabled for a few modules such as for this one).

cpinter, 2016-07-27:
If for SlicerIGT's use cases this "real-time behaviour" is better, and it is accepted that the modules in the extension don't work the same way as Slicer core, then auto-update is fine by me.

In this case adding None to the output combobox is enough.

tavaughan, 2016-08-03:
The GUI update will address this issue by allowing "None" in the output combobox.

tavaughan, 2016-08-03:
(And "None" will be selected by default)

tavaughan, 2016-11-09:
Should this be closed?

cpinter, 2016-11-09:
Has the above discussed None option been added?

tavaughan, 2016-11-09:
It was merged in this commit:
https://github.com/SlicerIGT/SlicerIGT/commit/c14531826551f9c0e6ae8e7a5aabaa64bc83ae95

cpinter, 2016-11-09:
Thanks! In that case I think this issue can be closed. I'm closing it now.

closed 2016-11-09 by cpinter