SlicerIGT/SlicerIGT #98: Add option to set name of the created model node

ungi, 2016-09-12:


tavaughan, 2017-06-22:
We could create new versions for each of the existing methods (CreateNeedle, etc) which also take strings:

`vtkMRMLModelNode* CreateNeedle( double length, double radius, double tipRadius, bool markers, std::string nodeName, vtkMRMLModelNode* modelNodeToUpdate = NULL );`

`vtkMRMLModelNode* CreateNeedle( double length, double radius, double tipRadius, bool markers, vtkMRMLModelNode* modelNodeToUpdate = NULL ) { /* call the version that takes a string, use a default name */ }`

But then the parameter list is getting rather long.

The alternative would be to create a parameter node for CreateModels, but this seems unnecessary to me given the simple nature of this particular module.

lassoan, 2017-06-22:
If nodes are created from another module then specifying the name is just one more line. What we need is to expose it on the GUI: have an Advanced section (collapsed by default) where you can specify a node name. There should be also a node selector in the advanced section to allow updating existing nodes. If a node is selected then this node name textbox should be disabled (the node selector will have a rename option anyway). If the user doesn't specify a custom node name (empty textbox) then the default name would be used. The user-specified name should be used as a base name (make the scene generate a unique name from that).