SlicerIGT/SlicerIGT #100: Make automatic flipping of needle orientation optional

mholden8, 2016-09-22:
When computing the pivot calibration or spin calibration, the automatic "flipping" of the needle should be made optional. This can be added as a parameter to the functions computing the pivot and spin calibrations (default behaviour should be to automatically flip).

Improve the documentation on this automatic flipping behaviour.

See discussion: https://github.com/SlicerIGT/SlicerIGT/commit/80106cce77376eb13399aa08d9da6ac862e8b532#commitcomment-19125336

lassoan, 2016-09-25:
@mholden8 Are you planning to implement this?

mholden8, 2016-09-26:
Yes, please see: https://github.com/mholden8/SlicerIGT/tree/auto-orient-pivot-spin-calibration. I asked @tavaughan to test it with his particular application before creating the pull request.

tavaughan, 2016-09-30:
Seems to work in CathNav.

mholden8, 2016-10-19:
Fixed in pull request #102.

closed 2016-10-19 by mholden8