SlicerIGT/SlicerIGT #107: Transform Fusion - Move the Timed Update timer elsewhere

tavaughan, 2016-11-09:
The timer doesn't belong in the widget, it would not work across multiple parameter nodes, updating multiple outputs

tavaughan, 2017-06-22:
Now that there is a real "Automatic Update", this feature does not seem to have much practical use. The options I see are:
- leave it as-is.
- remove it completely and all the issues associated with it.

Probably removing it is not much work, and could save future trouble.

lassoan, 2017-06-22:
What is this Timed Update timer?

tavaughan, 2017-06-22:
It is a QTimer, and the transform fusion output is updated on timeout

lassoan, 2017-06-22:
OK, if auto-update is now managed in the logic it's good and auto-update should be removed from GUI (and only manual update should be kept, the same way, preferably with the same checked button GUI as in MarkupsToModel).

tavaughan, 2017-06-22:
My intention is to make it behave the same way as in other modules.

tavaughan, 2017-10-24:
Timed update has been replaced with auto-update (event-based).

tavaughan, 2017-10-24:
No wait, I lied. It's still there.

closed 2017-10-27 by tavaughan