SlicerIGT/SlicerIGT #124: MarkupsToModel - stop undesired automatic fiducial placement

tavaughan, 2017-05-11:
Every time I change the input markups node, the interaction mode changes so that I'm placing fiducials with each click. In theory this is a convenience, but in practice it has caused me much more frustration than convenience. If I change the input node, and immediately want to rotate the view or click on something else in the viewers, I instead unintentionally place fiducials. This has happened to me dozens (if not hundreds) of times.

Furthermore, this behavior is inconsistent with the creation or change of markups node from the Markups module itself. If I create or change the markups node in that module, my interaction mode remains unchanged. This has led me to even more confusion/uncertainty whenever I browse markups in either module because the behavior is inconsistent.

I propose we remove this "feature" and let the user manually click on the "Place" button that is already next to the input selection box. This is consistent with the Markups module.

lassoan, 2017-05-12:
The behavior is controlled by "enterPlaceModeOnNodeChange" property of the qSlicerSimpleMarkupsWidget. If you find that the behavior is more annoying then helpful then set enterPlaceModeOnNodeChange to false (it is true by default).

tavaughan, 2017-05-12:
Thank you Andras, that sounds like a good plan. I do find this behavior unexpected, and sadly more annoying than helpful. I think it makes more sense for it to be disabled (at least by default).

tavaughan, 2017-05-19:
Whoops, somehow the commit disappeared from the list of commits. But it's back again: dd4052a

closed 2017-05-19 by tavaughan