SlicerIGT/SlicerIGT #138: Allow creation of model node from Collect Fiducials

tavaughan, 2017-10-02:
The model node would collect point data only.

tavaughan, 2017-10-13:
I've done this. But I have a problem. I add a point to the model as follows:
`void vtkSlicerCollectFiducialsLogic::AddPointToModel( vtkMRMLCollectFiducialsNode* collectFiducialsNode, vtkMRMLModelNode* outputModelNode, double* pointCoordinates )
{
  vtkSmartPointer< vtkPolyData > outputPolyData = outputModelNode->GetPolyData();
  if ( outputPolyData == NULL )
  {
    outputPolyData = vtkSmartPointer< vtkPolyData >::New();
    outputModelNode->SetAndObservePolyData( outputPolyData );
  }

  vtkSmartPointer< vtkPoints > outputPoints = outputPolyData->GetPoints();
  if ( outputPoints == NULL )
  {
    outputPoints = vtkSmartPointer< vtkPoints >::New();
    outputPolyData->SetPoints( outputPoints );
  }

  int numberOfPoints = ( int )outputPoints->GetNumberOfPoints();
  if ( collectFiducialsNode->GetCollectMode() == vtkMRMLCollectFiducialsNode::Automatic && numberOfPoints > 0 )
  {
    double previousCoordinates[ 3 ];
    outputPoints->GetPoint( numberOfPoints - 1, previousCoordinates );
    double distanceMm = sqrt( vtkMath::Distance2BetweenPoints( pointCoordinates, previousCoordinates ) );
    if ( distanceMm < collectFiducialsNode->GetMinimumDistanceMm() )
    {
      return;
    }
  }

  outputPoints->InsertNextPoint( pointCoordinates );
}`

But the vtkMRMLModelNode doesn't detect this (no Modified event is called). This gets to be a problem when I give this model as input to MarkupsToModel. I can get around this problem by manually calling:

`outputPolyData->Modified();
outputModelNode->Modified();`

But this doesn't seem right. There must be a cleaner way that does not involve manually calling Modified()?

lassoan, 2017-10-13:
In such cases, we typically add an observer to the VTK object in the MRML node. See for example how this->Plot object is observed: https://github.com/Slicer/Slicer/blob/master/Libs/MRML/Core/vtkMRMLPlotDataNode.cxx

Important:

1. The object setter method name is SetAndObserve...() not just Set...().

2. If the internal VTK object is expected to change frequently - which is true in our case - then instead of calling Modified(), it is recommended to invoke a new custom modified event to prevent overwhelming generic node observers (such as those that are there in GUI widgets) with low-level details. See vtkMRMLPlotDataNode::TableModifiedEvent in the referenced file.

lassoan, 2017-10-13:
I think the model node already observes the PolyData object, so calling outputPolyData->Modified()should be sufficient. Are you sure that outputModelNode->Modified() is needed?

outputPolyData->Modified() is necessary because most of the time you add hundreds of points to a polydata and change various other things and you don't want the rest of the application to react until you finished your update. You indicate that your update is finished by calling outputPolyData->Modified(). You might need to call Modified() for the PointData object instead (or in addition), but I'm not sure.

tavaughan, 2017-10-13:
I'll investigate tomorrow. It only worked when both Modified() events were called. Either one on its own did not work.

There are a few places in code that I'm a bit concerned about... it all works, but I'm sure there must be better ways to do certain things. I'll put together a list of places I'm concerned about. Alternatively, if you have ~15-30 min for a Skype call tomorrow, we can discuss a bit more interactively.

lassoan, 2017-10-13:
OK. You can call me between 9am and 3pm tomorrow.

tavaughan, 2017-10-13:
How to store node that is one of multiple different types:
vtkMRMLNode?
https://github.com/SlicerIGT/SlicerIGT/pull/141/files#diff-be722ca894ddf90d47a56408870edc65R83
https://github.com/SlicerIGT/SlicerMarkupsToModel/pull/7/files#diff-179e20bff449774a6a0cf004e1406d51R61

How to differentiate type:
https://github.com/SlicerIGT/SlicerIGT/pull/141/files#diff-d347780f4b8932dc19026e91223982ccR83
https://github.com/SlicerIGT/SlicerMarkupsToModel/pull/7/files#diff-b3c73f06a1311e927559f983d8b0dd1bR37
I saw just now there is a function called node->IsA( "vtkMRMLMarkupsFiducialNode" ). Is this the accepted way?

What about cases where only one type needs to be checked? Same idea?
https://github.com/SlicerIGT/SlicerMarkupsToModel/pull/7/files#diff-86097dd7a5edf72017565a5b42495299R434
https://github.com/SlicerIGT/SlicerMarkupsToModel/pull/7/files#diff-86097dd7a5edf72017565a5b42495299R465
https://github.com/SlicerIGT/SlicerMarkupsToModel/pull/7/files#diff-86097dd7a5edf72017565a5b42495299R568
https://github.com/SlicerIGT/SlicerMarkupsToModel/pull/7/files#diff-86097dd7a5edf72017565a5b42495299R590

GUI handling of markups widget when a markups node is not used as input. Right now there is an input node selector. If a markups node is selected, then the selector disappears and is replace with the markups widget and a button to 'deselect' the markups.

TODO: vtkNotUsed
https://github.com/SlicerIGT/SlicerMarkupsToModel/pull/7/files#diff-c456a52f978f9fb24d6d41756ae62b7fR325

TODO: name of 'markups' variable -> 'input'
https://github.com/SlicerIGT/SlicerMarkupsToModel/pull/7/files#diff-142e7dfc9a9abe5fa16ccab50d5dbe0aR162

tavaughan, 2017-10-13:
TODO: Create default display node on model node on creation.

TODO: Add vertex cells to model.

TODO:
invoke event invoked when poly data is changed
```events->InsertNextValue( vtkMRMLModelNode::MeshModifiedEvent );```

tavaughan, 2017-10-13:
TODO: Add qSlicerMarkupsPlaceWidget() instead of the 'simple' widget, hook it up with node selector when markups are selected

tavaughan, 2017-10-13:
TODO: Disable Label counter and base when a model output is selected

tavaughan, 2017-10-13:
TODO: Add button to remove one point, button to remove all points.

tavaughan, 2017-10-13:
Is there a way for me to use qSlicerMarkupsPlaceWidget from QtDesigner? If not, how should I go about this?

lassoan, 2017-10-13:
You need to add a designer plugin if you want the widget to show up in designer. It's just adding 2-3 files to Markups module widgets and update the CMakeLists.txt file. Just copy-paste work, probably taking less than 30 minutes. It would be nice if you could do it.

closed 2017-10-23 by tavaughan