SlicerIGT/SlicerIGT #145: Starting live reconstruction before scout scan results in unhelpful error message

Sunderlandkyl, 2017-10-19:
Starting the live volume reconstruction before performing a scout scan causes an error in Python:
~~~~
Traceback (most recent call last):
  File "C:/Users/kyles/AppData/Roaming/NA-MIC/Extensions-26478/SlicerIGT/lib/Slicer-4.7/qt-scripted-modules/PlusRemote.py", line 816, in onStartStopLiveReconstructionButtonClicked
    self.onStartReconstruction()
  File "C:/Users/kyles/AppData/Roaming/NA-MIC/Extensions-26478/SlicerIGT/lib/Slicer-4.7/qt-scripted-modules/PlusRemote.py", line 938, in onStartReconstruction
    self.logic.startVolumeReconstruction(self.linkInputSelector.currentNode().GetID(), self.volumeReconstructorIDSelector.currentText, self.outputSpacingValue, self.outputOriginValue, self.outputExtentValue, self.printCommandResponse, self.getLiveReconstructionOutputFilename(), self.liveOutputVolumeDeviceBox.text)
  File "C:/Users/kyles/AppData/Roaming/NA-MIC/Extensions-26478/SlicerIGT/lib/Slicer-4.7/qt-scripted-modules/PlusRemote.py", line 1435, in startVolumeReconstruction
    self.cmdStartVolumeReconstruction.SetCommandAttribute('OutputSpacing', '%f %f %f' % tuple(outputSpacing))
TypeError: 'int' object is not iterable
~~~~

As a novice SlicerIGT/Plus user, I'm assuming that a live volume reconstruction cannot be done until the ROI is determined. If that's the case, then it would probably help to have a more informative error message, or disable the live volume reconstruction button until the ROI has been set.

tavaughan, 2017-10-25:
@Sunderlandkyl can you try pulling the modified code below and trying that?

https://github.com/tavaughan/SlicerIGT/tree/plus-remote-live-reconstruction-warning

Sunderlandkyl, 2017-10-27:
Thanks Thomas, I don't get the error any more, and the new warning message is much more informative.

closed 2018-02-07 by Sunderlandkyl