SlicerIGT/SlicerIGT #152: Rename Transform Fusion?

tavaughan, 2017-10-27:
@lassoan We discussed renaming Transform Fusion to better reflect its new functionality. It can now concatenate transforms, and it can now separate translation or rotation information. Transform Processor (or something along those lines) was suggested in #44.

lassoan, 2017-10-27:
Transform Processor sounds good to me!

lassoan, 2017-10-27:
Can it also invert transforms?

tavaughan, 2017-10-27:
@lassoan Indirectly. You would need to compute full transform from parent to child, and store the result in another transform.

I could add a convenience mode to take node that stores transform to parent, and create a new transform node that stores inverted transform to parent?

closed 2017-11-03 by tavaughan