SlicerIGT/SlicerIGT #162: Allow user to define coordinate system for collected points

tavaughan, 2017-11-27:
Something like:
default = use the parent transform node (if it exists)
else = defined by the user, make sure to allow 'None' for world (Ras) coordinate system.

lassoan, 2017-11-27:
Set custom "none" text in the node selector widget to describe what happens when you select it.

tavaughan, 2018-03-07:
That sounds like a great idea. I could put "Ras" instead of "None". Do we have any examples that I could look at?

tavaughan, 2018-03-08:
Upon further revision of work, still need to:
- [ ] Change the CreateDefaultDisplayNodesForOutputNodes() so the output node is not capitalized
- [ ] Change SetAndObserve...NodeId to have "ID" capitalized
- [ ] Check to make sure the sampling/probe transform is not the same as anchor, this should at the very minimum log a warning. Probably should not be allowed to happen.

closed 2018-03-20 by tavaughan