SlicerIGT/SlicerIGT #169: Building against SlicerOpenIGTLink

tavaughan, 2018-03-14:
I'm running into issues building in Windows. I'm confused as to why CMake is asking for both a SlicerOpenIGTLink_DIR and OpenIGTLinkIF_DIR. I set SlicerOpenIGTLink_DIR to the inner-build folder of the SlicerOpenIGTLink project. But where am I supposed to set OpenIGTLinkIF_DIR?

lassoan, 2018-03-14:
Setting SlicerOpenIGTLink Directions should be enough. Have you built SlicerOpenIGTLink successfully in the same build mode (Debug / release) ? Which repository you've built SlicerOpenIGTLink from?

tavaughan, 2018-03-15:
Yes, I built SlicerOpenIGTLink in the same build mode. I built from: https://github.com/openigtlink/SlicerOpenIGTLink.git

I did not try generating + building after getting the CMake error message.

I tried reverting this commit and the error message went away: https://github.com/SlicerIGT/SlicerIGT/commit/3a2c7963c9dcd69b07af3fc29eb7a27aa0a061f4

lassoan, 2018-03-15:
Good catch. That seems to be an error!! I'll have another look and revert that change.

tavaughan, 2018-03-15:
I tried loading SlicerOpenIGTLink into slicer, I got an error message:

  Error(s):
    Cannot load library C:/D/SOIGTLD/inner-build/lib/Slicer-4.9/qt-loadable-modules/Debug/qSlicerOpenIGTLinkIFModule.dll: The specified module could not be found.

So I'm not sure anymore. I don't think the problem is with SlicerIGT since I'm not even trying to load it at the moment.

tavaughan, 2018-03-19:
Reverted in commit https://github.com/SlicerIGT/SlicerIGT/commit/8d7fbdc08d39ff6a14f9f38746c0fce7c88444e9

closed 2018-03-19 by tavaughan