SlicerIGT/SlicerIGT #171: Use QSignalBlocker to temporarily block signals

tavaughan, 2018-03-15:
Replace the old way (bool wasBlocked = ... do stuff ... setBlocked(wasBlocked) ).

Unless I am mistaken this is a Qt5 feature. So we should do this once Slicer and extensions have fully transitioned.

lassoan, 2018-03-15:
Slicer 4.8 is Qt4. Slicer 4.9 is Qt5, so you can use Qt5 features. It can wait, so it is enough for now that you added a ticket.

tavaughan, 2018-03-15:
Ok, thanks.

By the way, does Slicer 4.9 build yet with Qt5 on Linux? Is there a place I can follow the latest developments (apart from discourse)? I wouldn't mind adding preprocessor statements if I knew I could get a Linux build going.

cpinter, 2018-03-15:
I haven't seen a successful Linux factory build for a long time:
http://slicer.cdash.org/index.php?project=SlicerPreview

lassoan, 2018-03-15:
Linux build is almost working. It should be ready within a few days, if not already working. If you have any particular issue, post it at https://discourse.slicer.org/t/build-3dslicer-under-ubuntu-16-04/2322/5