SlicerIGT/SlicerIGT #186: OpenIGTLink connection to an Atracsys fusionTrack500

dimitri80, 2018-12-05:
Hello,
I am trying to connect an Atracsys fusionTrack500 tracker to 3D Slicer 4.10.0. so that I can create a  sequence metafile that tracks the StylusTip.
I successfully can start the Plus Server, and connect via OpenIGTLinkIF without error, but when I expand the Scene in my I/O Configuration, both IN and OUT are blank.  Below is my config file as well as error log, showing the connection.  Any help would be greatly appreciated. 

[image]
[image]
[atracsys.txt](https://github.com/SlicerIGT/SlicerIGT/files/2649594/atracsys.txt)

dimitri80, 2018-12-05:
Hi, here is the log file for PLUS, I am not seeing any error, but i am very new to this system so I can be wrong.  Thank you for your expertise
[atracsys.txt](https://github.com/SlicerIGT/SlicerIGT/files/2650120/atracsys.txt)
.

markasselin, 2018-12-05:
@dimitri80 Have you defined your marker ini file correctly?

markasselin, 2018-12-05:
Since you're using passive markers only, set the ActiveMarkerPairingTimeSec to "0". And in you PlusOpenIGTLinkServer could you please set SendValidTransformsOnly to "FALSE". Alternatively, you can try increasing the MaxMeanRegistrationErrorMm attribute.

dimitri80, 2018-12-05:
@markasselin
You nailed it, thank you.

lassoan, 2018-12-05:
Great to see that it's solved. @markasselin could you please add a note about this to https://github.com/PlusToolkit/PlusLib/blob/master/src/Documentation/UserManual/DeviceAtracsys.dox to help future users find this useful tip more easily? Thanks!

closed 2019-01-18 by dimitri80