SlicerIGT/SlicerIGT #199: What to do with ImageToProbe transform ?

AurelieSar, 2020-08-24:
Hi,
I switched from the PlusToolkit forum to this forum since my questions are now more directed to the use of SlicerIGT. 

My objective is to record 3D ultrasound and to do first an offline reconstruction, and if possible then a live reconstruction. 
I am using an Optitrack motion capture system (Flex 13) with an SuperShearingImaging ultrasound which video feed arrives to the computer through an USB frame grabber (CAM LINK). 

After adjusting the "ImageToReference" matrix to select only the ultrasound image, I succeeded in doing a temporal calibration and stylus calibration using fCal.
After saving my updated configuration file (see attached), I launch the PlusServer and then 3D Slicer. I add my two connectors (video feed and tracking). 

**Question 1 :** 
I had an hard time doing the spatial calibration, since I had to:
- record the sequence of the spatial calibration using Plus Remote
- re-upload this data into 3D Slicer using Sequence
- rearrange all the different data to be able to visualize it in the 2D and 3D windows
- follow the tutorial using the fiducials. 
Is there an easier way to do the spatial calibration (using Slicer 3D, I want to test without the phantom first), maybe "live" or just by freezing the images ? or is it why the calibration using the phantom has been developed to be able to do it live ?

**Question 2 :** 
I finally had my ImageToProbe calibration matrix (see attached). However I do not know what to do with it ? I tried one transform hierarchy (see attached), however the picture from the ultrasound diminished a lot and became very very small in both the 2D and 3D views. Indeed the scaling factor in my ImageToProbe is 0.01 .. Is that usual ? Although the projected ultrasound image followed nicely the probe marker needle in the 3D view. Am I missing something ? I cannot find the next steps after spatial calibration in the tutorials .. 

[image]
[image]
[image]
[image]
[image]

Thanks a lot in advance,
Cheers,
Aurélie

Sunderlandkyl, 2020-08-24:
Which tutorial were you following for ultrasound calibration, U-31?

You can record the image+transforms in the Sequences module in Slicer, rather than recording using PlusRemote.

The coordinate system in your config file is ImageToReference. I think that it should be ImageToProbe?

If your output image uses "Reference" embedded transform frame, then you do not need to apply any other transforms to it in Slicer. If you use the "Image" embedded transform frame, you can set up your transform hierarchy like this:
- ProbeToReference
  - ImageToProbe
    - Image_Image

ungi, 2020-08-24:
There is a confusion, because in early systems, we sent ImageToReference from PLUS to Slicer. So when we computed ImageToProbe, we had to add it to the PLUS config file, so PLUS can compute ImageToReference=ProbeToReference*ImageToProbe.
There are still lots of PLUS config file examples for sending ImageToReference from the PLUS to Slicer.

However, Slicer has improved a lot in the past years, so now the preferred way is to send Image in the Image reference coordinate system (untransformed Image) from PLUS to Slicer. Additionally, you need to send ProbeToReference from PLUS to Slicer too.
Slicer now has the Sequences module to save time series, and U-31 teaches how to compute ImageToProbe, so you can set up the transform hierarchy @Sunderlandkyl just mentioned in the previous comment. Today, this is the preferred way of calibrating, recording, and 3D-reconstruction for tracked ultrasound images. Today, I record data in PLUS to mha files is used for testing, but not for any application.

You can certainly pause (uncheck active) in the OpenIGTLinkIF module to pause the tracked ultrasound at any point in time to record Image points and a Probe points for calibration. I would still recommend using Sequences to record all the calibration data and archive it. That way, you can always go back to the original sequence recording, if you doubt the calibration result later.

ungi, 2020-08-24:
By the way, use Slicer discourse for these discussions please. There is an IGT category in discourse.

AurelieSar, 2020-08-24:
Thank you a lot for your answers, and sorry I did not see the discourse website. Can I finish with that point here or do you prefer I switch now to the discourse website ?

My follow-up questions are :
1. I used the matrix from "image" to "reference" in the config file in order to resize and translate the image from the ultrasound and make it visible in the red box of Slicer 3D. How could I do that without the matrix since now I send the untransformed image ? it is in the config file or directly in slicer ?
2. I don't quite get the difference between "Tracker" and "Reference"... what transforms are necessary to send to Slicer ? I never find the same in the example config files ... Could it be possible to have a "recent" configuration files of a setup using motion capture and ultrasound to help me ? All the config files I found still use "Image to Reference" and it keeps confusing me ..

ungi, 2020-08-24:
1. You can always resize the slice field of view to fit the image content with this button:
[image]

2. "Tracker" usually refers to the camera coordinate system, which is defined by OptiTrack (or whatever tracker system you are using). Although "Reference" is not absolutely necessary, we usually add a tracked object that does not move relative to the patient. Reference is useful when the tracker cameras need to move, but you have already localized your patient.

AurelieSar, 2020-08-25:
Thank you so much, all is way clearer now !
Aurélie