SlicerIGT/SlicerIGT #201: Post install Slicer console errors

msimonc, 2021-04-25:
New Install.  I have Qt 5.15.2 and Slicer 4.13, thanks
```
    The plugin 'C:/D/S4D/Slicer-build/NA-MIC/Extensions-29848/SlicerIGT/lib/Slicer-4.13/qt-loadable-modules
/qSlicerBreachWarningModule.dll' uses incompatible Qt library. (Cannot mix debug and release libraries.)
  Error(s):
    The plugin 'C:/D/S4D/Slicer-build/NA-MIC/Extensions-29848/SlicerIGT/lib/Slicer-4.13/qt-loadable-modules
/qSlicerCollectPointsModule.dll' uses incompatible Qt library. (Cannot mix debug and release libraries.)
  Error(s):
    The plugin 'C:/D/S4D/Slicer-build/NA-MIC/Extensions-29848/SlicerIGT/lib/Slicer-4.13/qt-loadable-modules
/qSlicerCreateModelsModule.dll' uses incompatible Qt library. (Cannot mix debug and release libraries.)
  Error(s):
    The plugin 'C:/D/S4D/Slicer-build/NA-MIC/Extensions-29848/SlicerIGT/lib/Slicer-4.13/qt-loadable-modules
/qSlicerFiducialRegistrationWizardModule.dll' uses incompatible Qt library. (Cannot mix debug and release libraries.)
  Error(s):
    The plugin 'C:/D/S4D/Slicer-build/NA-MIC/Extensions-29848/SlicerIGT/lib/Slicer-4.13/qt-loadable-modules
/qSlicerPathExplorerModule.dll' uses incompatible Qt library. (Cannot mix debug and release libraries.)
  Error(s):
    The plugin 'C:/D/S4D/Slicer-build/NA-MIC/Extensions-29848/SlicerIGT/lib/Slicer-4.13/qt-loadable-modules
/qSlicerPivotCalibrationModule.dll' uses incompatible Qt library. (Cannot mix debug and release libraries.)
  Error(s):
    The plugin 'C:/D/S4D/Slicer-build/NA-MIC/Extensions-29848/SlicerIGT/lib/Slicer-4.13/qt-loadable-modules
/qSlicerTransformProcessorModule.dll' uses incompatible Qt library. (Cannot mix debug and release libraries.)
  Error(s):
    The plugin 'C:/D/S4D/Slicer-build/NA-MIC/Extensions-29848/SlicerIGT/lib/Slicer-4.13/qt-loadable-modules
/qSlicerUltrasoundSnapshotsModule.dll' uses incompatible Qt library. (Cannot mix debug and release libraries.)
  Error(s):
    The plugin 'C:/D/S4D/Slicer-build/NA-MIC/Extensions-29848/SlicerIGT/lib/Slicer-4.13/qt-loadable-modules
/qSlicerVolumeResliceDriverModule.dll' uses incompatible Qt library. (Cannot mix debug and release libraries.)
  Error(s):
    The plugin 'C:/D/S4D/Slicer-build/NA-MIC/Extensions-29848/SlicerIGT/lib/Slicer-4.13/qt-loadable-modules
/qSlicerWatchdogModule.dll' uses incompatible Qt library. (Cannot mix debug and release libraries.)
```

lassoan, 2021-04-25:
If you install a pre-built Qt (and not build it yourself) then you'll only get a release-mode-built Qt designer. That is not compatible with debug-mode-built designer plugins.

msimonc, 2021-04-26:
I built Release mode and used Extension Manager to install SlicerIGT, SlicerIGSIO, SlicerOpenIGTLink and SlicerJupyter.
Errors on restart:
(without error, these files had not been installed)
```
Error(s):
   Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerIGT\lib\Slicer-4.13\qt-loadable-modules\qSlicerBreachWarningModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerIGT\lib\Slicer-4.13\qt-loadable-modules\qSlicerCollectPointsModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerIGT\lib\Slicer-4.13\qt-loadable-modules\qSlicerCreateModelsModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerIGT\lib\Slicer-4.13\qt-loadable-modules\qSlicerFiducialRegistrationWizardModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerIGT\lib\Slicer-4.13\qt-loadable-modules\qSlicerPathExplorerModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerIGT\lib\Slicer-4.13\qt-loadable-modules\qSlicerPivotCalibrationModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerIGT\lib\Slicer-4.13\qt-loadable-modules\qSlicerTransformProcessorModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerIGT\lib\Slicer-4.13\qt-loadable-modules\qSlicerUltrasoundSnapshotsModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerIGT\lib\Slicer-4.13\qt-loadable-modules\qSlicerVolumeResliceDriverModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerIGT\lib\Slicer-4.13\qt-loadable-modules\qSlicerWatchdogModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerIGSIO\lib\Slicer-4.13\qt-loadable-modules\qSlicerMetafileImporterModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerIGSIO\lib\Slicer-4.13\qt-loadable-modules\qSlicerSequenceIOModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerIGSIO\lib\Slicer-4.13\qt-loadable-modules\qSlicerVideoUtilModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerIGSIO\lib\Slicer-4.13\qt-loadable-modules\qSlicerVolumeReconstructionModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerJupyter\lib\Slicer-4.13\qt-loadable-modules\qSlicerJupyterKernelModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerOpenIGTLink\lib\Slicer-4.13\qt-loadable-modules\qSlicerOpenIGTLinkIFModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerOpenIGTLink\lib\Slicer-4.13\qt-loadable-modules\qSlicerOpenIGTLinkRemoteModule.dll: The specified module could not be found.
  Error(s):
    Cannot load library C:\D\S4R\Slicer-build\NA-MIC\Extensions-29848\SlicerOpenIGTLink\lib\Slicer-4.13\qt-loadable-modules\qSlicerPlusRemoteModule.dll: The specified module could not be found.
```

lassoan, 2021-04-27:
Do you have any third-party antivirus software installed?

msimonc, 2021-04-27:
My mistake, those files do exist.  Same errors.  No antivirus software installed (dev only PC).

lassoan, 2021-04-27:
Please check in Windows Event Viewer if there is information about why the DLL failed to load. Typically it is because of a required DLL is not found, and the name of the DLL is logged.

ML-lilith, 2023-07-11:
The same errors happeded on my desktop today. 
windows. didn't install any software recently. The 3D Slicer opened yesterday. 
 Error(s):
    The plugin 'C:/D/S4D/Slicer-build/slicer.org/Extensions-31764/SlicerJupyter/lib/Slicer-5.3/qt-loadable-modules/qSlicerJupyterKernelModule.dll' uses incompatible Qt library. (Cannot mix debug and release libraries.) 
The files asked are existed in the path.   Do you solve this problem?

lassoan, 2023-07-11:
Make sure you selected the correct kernel in the notebook (i.e., select the debug-mode kernel if you use a Slicer that you have built in debug mode).

ML-lilith, 2023-07-11:
> Make sure you selected the correct kernel in the notebook (i.e., select the debug-mode kernel if you use a Slicer that you have built in debug mode).

Thanks for your kind reply. But I didn't use the SlicerJupyter recently. This is an extension i installed a month ago.  Without any reason, I can't open the debug version of 3D slicer Today. This errors showed on the popup-cmd when I try to open the 3D slicer Today. I tryed to delete the whole extension packages. But the 3D slicer still can not open. 
like this, 
[image]

lassoan, 2023-07-12:
Have you updated 3D Slicer source code recently?

ML-lilith, 2023-07-12:
I do add an extension under the path of "C:\D\S4D\Slicer-build\Extensions". Does that count as an update to the source code? And I do try to verify the some python files on local 3D slicer.

ML-lilith, 2023-07-12:
> I do add an extension under the path of "C:\D\S4D\Slicer-build\Extensions". Does that count as an update to the source code? And I do try to verify the some python files on local 3D slicer.

I solved the prolem. @lassoan . I double-check the python files for my extension that are contained within the slicer source code, as you mentioned. The 3D slicer may now be launched after I deleted a widget.py that used PyQt5.  So let's ask another question: Can we test our own extensions without putting them inside the slicer root path by simply keeping them in another path and testing them in the slicer-terminal?

lassoan, 2023-07-12:
Yes, sure, you just need to add the folders where your modules are to "Additional module paths" in Slicer application settings - see https://slicer.readthedocs.io/en/latest/developer_guide/extensions.html#run-slicer-with-your-custom-modules

closed 2023-07-12 by lassoan