SlicerIGT/SlicerIGT #212: Fiducial Registration Wizard: Undo via Ctrl+Z

henrykrumb, 2023-07-01:
It would be great to have an "undo-stack" in the fiducial registration wizard for placing the "From" and "To" markups, and a Ctrl+Z keybinding to revert the placement of the last fiducial.

Background: We often experienced problems with accidentally moving the fiducials when working on a laptop with a touchpad (OR space is confined and we couldn't really use a mouse). When one fiducial is moved by accident, it needs to be re-taken.

henrykrumb, 2023-07-01:
Addendum: I don't know for sure if this would need a change in SlicerIGT/FiducialRegistrationWizard or in the Slicer UI that is responsible for placing markups.

lassoan, 2023-07-01:
Thanks for the suggestion. This will be a Slicer core feature, so we will not track it here. 

Please vote for this feature and describe your use case here: https://discourse.slicer.org/t/is-it-possible-to-add-a-global-undo-button/16859

closed 2023-07-01 by lassoan