SlicerIGT/SlicerIGT #224: Cannot start Guidelet due to i18n strings (name '_' is not defined)

mcfly3001, 2024-11-22:
I have a developed a small Guidelet which is based on the SlicerIGT Guidelet. After updating to a newer version of Slicer (4.6.2), I cannot start the Guidelet anymore because it crashes in the `Ultrasound.py` file with the following error:

```
name '_' is not defined
```

I assume that this is related to [this commit](https://github.com/SlicerIGT/SlicerIGT/commit/b02db568ae34373d7b30fb196d07a302b0317350) where i18n was introduced. The `Ultrasound.py` is not importing the `_` as for example it is done in `Guidelet.py`:

```python
from slicer.i18n import tr as _
```

Is there a missconfiguration on my side when I am trying to start my Guildelet or is simply this line missing in `Ultrasound.py`?

lassoan, 2024-11-23:
You need to use the current extension version with current Slicer version. Internationalization was introduced in Slicer-5.6, the `_()` translation function is not available in earlier Slicer versions.

mcfly3001, 2024-11-23:
I am using Slicer 5.6.2 with the SlicerIGT extension that should suit to this version as I typically install everything fresh with the extensions provided from the extension downloader.
As mentioned, the `Guidelet.py` does not throw an error because i18n is imported there and usage of `_` works. Only as soon as `setupPanel` in the `Ultrasound.py` is called, the exception is raised.

lassoan, 2024-11-25:
OK, I understand it now. You need to add the missing `from slicer.i18n import tr as _` import. When you have fixed the issues on your computer then please submit a pull request with the necessary changes. Thank you!

mcfly3001, 2024-11-26:
Thanks for your feedback. I just was not sure if I missed something. The solution was simply to also import i18n as `_`. The PR for this is [here](https://github.com/SlicerIGT/SlicerIGT/pull/225).

lassoan, 2024-11-26:
Great, thank you, your fix has now been integrated:

https://github.com/SlicerIGT/SlicerIGT/commit/d291414619e6f204476f4c6c66dd682d1e2e3026

closed 2024-11-26 by lassoan